bluenviron/gomavlib #3: No error generated for failed client connections.

ben-crowhurst, 2019-12-14:
The following example will fail to reach the TCP Client endpoint, if none exists, and will forever wait on `node.Events()`.  There appears to be no way of detecting the failure, is this by design?
``` golang
	node, err := gomavlib.NewNode(gomavlib.NodeConf{
		Endpoints: []gomavlib.EndpointConf{
			gomavlib.EndpointTcpClient{"127.0.0.1:5600"},
		},
		Dialect:     ardupilotmega.Dialect,
		OutSystemId: 10,
	})
	if err != nil {
		panic(err)
	}
	defer node.Close()

	for evt := range node.Events() {
		if frm, ok := evt.(*gomavlib.EventFrame); ok {
			fmt.Printf("received: id=%d, %+v\n", frm.Message().GetId(), frm.Message())
		}
	}
```

aler9, 2019-12-14:
Hi this is by design, but let me explain why:
Mavlink is used to communicate with remote devices, so connection drops and failures are not errors, but a common phenomenon and must be handled. Furthermore, it's also possible that the device you're trying reach is turned off when the script is running, or is already in the initialization phase.

So: a `Node{}` doesn't return errors, since a connection drop is not an error, instead it:
* ignores parsing errors
* periodicaly tries to reconnect to the device when the connection drops
An Event is returned when a device appears and another Event when a device disappears, you can use them to detect when the device is online and when it is not:
```go
	for evt := range node.Events() {
		switch ee := evt.(type) {
		case *gomavlib.EventFrame:
			fmt.Printf("frame received: %v\n", ee)

		case *gomavlib.EventChannelOpen:
			fmt.Printf("device appeared: %v\n", ee)

		case *gomavlib.EventChannelClose:
			fmt.Printf("device disappeared: %v\n", ee)
		}
	}
```

The reconnection procedure, in case of a client, is here:
https://github.com/aler9/gomavlib/blob/master/endpoint_client.go#L115

If you don't like this behavior and want to directly handle errors, you can use a `Parser{}`:
```go
conn, err := net.Dial("tcp", "127.0.0.1:5600")
if err != nil {
    panic(err)
}
defer conn.Close()

parser, err := gomavlib.NewParser(gomavlib.ParserConf{
    Reader:      conn,
    Writer:      conn,
    Dialect:     ardupilotmega.Dialect,
    OutSystemId: 10,
})
if err != nil {
    panic(err)
}

for {
    frame, err := parser.Read()
    if err != nil {
        panic(err)
    }

    fmt.Printf("decoded: %+v\n", frame)
}
```

Keep in mind that there could be errors both in the connection phase and in the parsing phase, and you have to handle them.

ben-crowhurst, 2019-12-15:
No that is perfect!  Saves me having to implement the logic.  Thank you!

closed 2019-12-15 by ben-crowhurst