bluenviron/gomavlib #6: Access to internal message fields?

justinclift, 2020-06-27:
What's the correct way to access fields in the EventFrame Message structure?

At the moment (just getting my toes wet :wink:), I'm trying a simplistic approach of switching on the message ID (eg `case 0` for the heartbeat, then casting to the matching type. eg:

```
if frm, ok := evt.(*gomavlib.EventFrame); ok { // <-- from the example code ;)

	if frm.SystemId() != 1 { // Only decode messages from the Ground Control Station

		switch frm.Message().GetId() {
		case 0:
			fmt.Println("Heartbeat from Ground Control Station")
			fmt.Printf("MAVLink version: %d\n", frm.Frame.GetVersion())

			decoded, ok := frm.Message().(*ardupilotmega.MessageHeartbeat)
			if !ok {
				fmt.Println("Didn't get it right ;)")
				syscall.Exit(255)
			}
			fmt.Printf("Decoded MAV type: %v\n", decoded.Type)
			fmt.Printf("Decoded System status: %v\n", decoded.SystemStatus)
			fmt.Printf("Decoded MAVLink version: %d\n", decoded.MavlinkVersion)
		default:
			// ...
		}
```

Pretty sure I'm doing something obviously silly here though, as the output looks ok for some fields but the MAVLink version field doesn't match the version reported by `GetVersion()`. eg:

```
Heartbeat from Ground Control Station
MAVLink version: 1
Decoded MAV type: MAV_TYPE_GCS
Decoded System status: MAV_STATE_ACTIVE
Decoded MAVLink version: 3
```

The "MAVLink version" and "Decoded MAVLink version" there should match, as they're coming from the same message though accessed in two different ways.

Any ideas?

aler9, 2020-06-27:
Hi, there are multiple things that must be addressed separately:

* The MavlinkVersion of the heartbeat field is not the Mavlink version, it's an uint8 and is always 3 (this is part of the specification). The Mavlink version influences the magic byte at the beginning of each frame, and the library is able to decode both Mavlink 1.0 and 2.0. If you want to discover the Mavlink version of the frames you are receiving, use
   ```
   frm.Frame.GetVersion()
   ```
   and you'll get either `gomavlib.V2` or `gomavlib.V1`

* there's no need to perform a manual casting to access message fields, it's enough to use the switch statement with `(type)` and you'll get a variable which is already an `*ardupilotmega.MessageHeartbeat`:
```go
	for evt := range node.Events() {
		if frm, ok := evt.(*gomavlib.EventFrame); ok {
			switch msg := frm.Message().(type) {
			case *ardupilotmega.MessageHeartbeat:
				fmt.Println("received heartbeat:", msg.Type, msg.SystemStatus, msg.MavlinkVersion)
			}
		}
	}
```

justinclift, 2020-06-27:
Awesome, thanks heaps! :smile:

closed 2020-06-27 by justinclift