bluenviron/gomavlib #7: Commands

dvtkrlbs, 2020-07-18:
I want to use this library for my drone project but i can't figure out on how to create commands. Is it possible to do ?

dvtkrlbs, 2020-07-18:
https://mavlink.io/en/services/command.html here is the microservice documentation for it and i was also wondering if you have plans to add support for some of these services ?

aler9, 2020-07-30:
Hello, microservices are a set of features built upon Mavlink messages. Gomavlib currently implements all Mavlink messages but only the heartbeat service, while the other services are not implemented, since they're very hardware-dependent.

Nonetheless, nothing forbids you to send a command, which is just a simple message, to a drone, without waiting the ACK (the service allows to send commands and wait for acks).

Just find the message you're interested in from the documentation of the ardupilotmega dialect:
https://pkg.go.dev/github.com/aler9/gomavlib@v0.0.0-20200728030502-33a7a55313b3/dialects/ardupilotmega?tab=doc

In this case is MessageCommandInt, documented here:
https://pkg.go.dev/github.com/aler9/gomavlib@v0.0.0-20200728030502-33a7a55313b3/dialects/ardupilotmega?tab=doc#MessageCommandInt

then send it after you've received the UAV heartbeat, like in the [message_write](https://github.com/aler9/gomavlib/blob/master/examples/message_write.go) example (remove the fields you don't need):
```go
node.WriteMessageAll(&ardupilotmega.MessageCommandInt{
    TargetSystem: 0,
    TargetComponent: 0,
    Frame: 0,
    Command: 0,
    Current: 0,
    Autocontinue: 0,
    Param1: 0,
    Param2: 0,
    Param3: 0,
    Param4: 0,
    X: 0,
    Y: 0,
    Z: 0,
})
```

closed 2020-08-02 by dvtkrlbs