bluenviron/gomavlib #10: Help on getting telemetry stream

davebest, 2021-08-12:
I'm using ArduSub and am creating a stand-alone program to retrieve the depth of the ROV and output it as an NMEA-0183 message.  I would like send a PARAM_REQUEST_READ message and receive the stream over UDP.

Looking at the examples I'm not sure if I should connect as a udp-client, or udp-broadcast, since I'd like the stream to be directed rather than broadcast.
I have studied the examples and looked a bit into the source code, but I'm afraid I'm stumped.  I have tried to include the message to dialect.Messages, with no success:

    msgs := []msg.Message{}
    msgs = append(msgs, &common.MessageParamRequestRead{1, 1, "", 6})
    dialect := ardupilotmega.Dialect
    dialect.Messages = msgs
    node, err := gomavlib.NewNode(gomavlib.NodeConf{
        Endpoints: []gomavlib.EndpointConf{
            gomavlib.EndpointUDPBroadcast{
                BroadcastAddress: "192.168.2.255:14550",
                LocalAddress:     ":14550",
            },
        },
        Dialect:                dialect,
        OutVersion:             gomavlib.V2, // change to V1 if you're unable to communicate with the target
        OutSystemID:            10,
        StreamRequestEnable:    true,
        StreamRequestFrequency: 4,
    })

Any help would be appreciated.  A new example showing this common requirement might be helpful for all.

aler9, 2021-08-12:
Hello, the `common` dialect is already included in the `ardupilotmega` dialect, so there's no need to add `MessageParamRequestRead` again:

https://github.com/mavlink/mavlink/blob/1eac49b6d8c00d4051ff6713e5c49088cf75f7c5/message_definitions/v1.0/ardupilotmega.xml#L3

Regarding the communication issue, it strongly depends on what you're communicating with. The target must explicitly have a open UDP port that accepts Mavlink. UDP client usually works, there's no need of using UDP broadcast. The only advice i can give is to try to read before trying to send; once you've successfully received a message, you can easily send one back.

davebest, 2021-08-16:
Thank you.  I added  "--out udpin=192.168.2.2:14555" to the mavproxy config on the ROV and am able to connect to it with the UDP client.

In my setup mavproxy on the ROV is broadcasting heartbeats on 192.168.2.255:14550.  How can I can listen for a heartbeat message and get the sender ip and port from it, so that I can then do a UDP client connect to it (similar to how QGroundControl does it)?

aler9, 2021-08-18:
> In my setup mavproxy on the ROV is broadcasting heartbeats on 192.168.2.255:14550.  How can I can listen for a heartbeat message and get the sender ip and port from it, so that I can then do a UDP client connect to it (similar to how QGroundControl does it)?

This is a complex technique that unfortunately you'd have to implement by scratch if you want to use the library, but i can suggest a simpler approach:

1) replace mavproxy with https://github.com/aler9/mavp2p
2) mavp2p allows you to choose exactly what protocol you want to use to communicate with the ROV through serial. Choose a protocol and use it for direct communication, without using a third one (broadcast)