bluenviron/gomavlib #16: crash on github.com/aler9/gomavlib/pkg/frame.(*V2Frame).Decode

tubzby, 2021-11-25:
I'm using this lib to test a px4 UAV, somehow I hit on this crash, two times, no clue yet.

```
Nov 25 14:57:16 gdcx uavm[208678]: panic: runtime error: slice bounds out of range [:920] with capacity 512
Nov 25 14:57:16 gdcx uavm[208678]: goroutine 1923 [running]:
Nov 25 14:57:16 gdcx uavm[208678]: bufio.(*Reader).Read(0xc0004cfaa0, {0xc0004ae7e0, 0xc0003afe08, 0x44c532})
Nov 25 14:57:16 gdcx uavm[208678]:         /usr/local/go/src/bufio/bufio.go:238 +0x2ed
Nov 25 14:57:16 gdcx uavm[208678]: io.ReadAtLeast({0xd07100, 0xc0004cfaa0}, {0xc0004ae7e0, 0x20, 0x20}, 0x20)
Nov 25 14:57:16 gdcx uavm[208678]:         /usr/local/go/src/io/io.go:328 +0x9a
Nov 25 14:57:16 gdcx uavm[208678]: io.ReadFull(...)
Nov 25 14:57:16 gdcx uavm[208678]:         /usr/local/go/src/io/io.go:347
Nov 25 14:57:16 gdcx uavm[208678]: github.com/aler9/gomavlib/pkg/frame.(*V2Frame).Decode(0xc00031c600, 0x3)
Nov 25 14:57:16 gdcx uavm[208678]:         /Users/liuzhiping/go/pkg/mod/github.com/aler9/gomavlib@v0.0.0-20211028051547-c91b43b554d6/pkg/frame/v2.go:141 +0x1f3
Nov 25 14:57:16 gdcx uavm[208678]: github.com/aler9/gomavlib/pkg/transceiver.(*Transceiver).Read(0xc000678b80)
Nov 25 14:57:16 gdcx uavm[208678]:         /Users/liuzhiping/go/pkg/mod/github.com/aler9/gomavlib@v0.0.0-20211028051547-c91b43b554d6/pkg/transceiver/transceiver.go:130 +0x79
Nov 25 14:57:16 gdcx uavm[208678]: github.com/aler9/gomavlib.(*Channel).run.func1()
Nov 25 14:57:16 gdcx uavm[208678]:         /Users/liuzhiping/go/pkg/mod/github.com/aler9/gomavlib@v0.0.0-20211028051547-c91b43b554d6/channel.go:93 +0xbc
Nov 25 14:57:16 gdcx uavm[208678]: created by github.com/aler9/gomavlib.(*Channel).run
Nov 25 14:57:16 gdcx uavm[208678]:         /Users/liuzhiping/go/pkg/mod/github.com/aler9/gomavlib@v0.0.0-20211028051547-c91b43b554d6/channel.go:85 +0xef
```

tubzby, 2021-11-26:
Hi @aler9 , thanks for your excellent work! 
I might have found a bug which I first post it here: https://github.com/golang/go/issues/49795

aler9, 2021-12-08:
This have been fixed, thanks for the report, but i have a couple of things to say:

* next time please share at least the node configuration, otherwise it's nearly impossible to debug the issue.
* the Golang core library is tested with fuzzy tests and by a huge community, it surely wasn't a Go issue.
* since you found the solution, you'd better have reported it back.

closed 2021-12-08 by aler9