bluenviron/gomavlib #17: Trouble with MessageParamExtRequestList

xinnad, 2021-12-03:
I'm trying to get the full parameter list over mavlink with 
```
node.WriteMessageAll(&common.MessageParamExtRequestList{
		TargetSystem: d.Sysid,
		TargetComponent: d.CompId,
	})
```
I think I should receive the list of params as a response but I don't see anything with:
```
node, err := gomavlib.NewNode(gomavlib.NodeConf{
		Endpoints: []gomavlib.EndpointConf{
			gomavlib.EndpointUDPServer{Address: url},
		},
		Dialect:     common.Dialect,
		OutVersion:  gomavlib.V1, // change to V1 if you're unable to communicate with the target
		OutSystemID: 254,
	})
	if err != nil {
		panic(err)
	}

	defer node.Close()

	// Set LLA in our global drone fleet struct
	for evt := range node.Events() {
		if frm, ok := evt.(*gomavlib.EventFrame); ok {
			switch msg := frm.Message().(type) {
			
			case *common.MessageParamExtValue:
				fmt.Printf("  Param Val: %v\n", msg)

```
I can see responses from MAV_CMD items  with `*common.MessageCommandAck` but not from the param list. 

I didn't see any examples that included parameter requests with responses and was curious if anyone had something. 

Thanks for your help!

aler9, 2021-12-08:
Hello, if you can see responses from `MAV_CMD` it means that communication is working properly.

If you can't see responses from `MessageParamExtRequestList`, it means that your drone doesn't support the extened parameter protocol.

xinnad, 2021-12-09:
Confirmed I was using the wrong command. `MessageParamRequestList` returned all params as `MessageParamValue` like I'd expect.

closed 2021-12-09 by xinnad