bluenviron/gomavlib #18: Int32 param fails but Real32 works

xinnad, 2022-01-04:
I'm having a problem uploading int32 parameters to a PX4 flight controller. Real32 params work correctly.

- I can send an int32 param upload message and receive a param value ACK message
- The value of the ACK matches the value of the upload
- When I query the flight controller directly I get huge values except for uploading zero (i.e. I upload an integer 1 and the flight controller reports 1065353216). 
 
I can upload params successfully with QGroundControl so I'm pretty sure the problem lies somewhere in gomavlib or how I'm using it. If you have any advice where to look I'd appreciate it.

Thanks for your help!

xinnad, 2022-01-04:
It looks like the issue comes from the way the flight controller casts the parameter value. The line in question is `s->val.i = *(int32_t *)val;` The equivalent line for a float param is `s->val.f = *(float *)val;` which works correctly. It seems like even though the param set message always sends a float value, the binary data contained in that float value is directly cast as whatever datatype the PARAM_TYPE_XXX says, in this case an int32. 

A potential solution is:
- gomavlib casts the data to an int32 (or other MAV_PARAM_TYPE) binary representation
- casts that binary representation to a float*
- sets the float value in the param set message to that binary representation

Does this seem like a reasonable approach? I'm not sure if this is an issue on other flight controllers or if it's only an issue for PX4.

xinnad, 2022-01-04:
I think I found a solution/workaround. It might be worthwhile to implement this in the gomavlib library so a user doesn't need to worry about casting datatypes. I used a standard type cast to convert the floating point value to an int, converted the int to its binary form with the `encoding/binary` package, and then wrote the resulting binary representation back to the address of my floating point parameter value. Things seem to work now.

xinnad, 2022-01-20:
Seems like this is the way the message is intended to work. The solution I proposed has been tested on hardware and seems to be reliable.

closed 2022-01-20 by xinnad

aler9, 2022-01-27:
Hello, dialects are automatically generated from definitions, therefore this issue is common to all programming languages. At the moment there aren't much utilities in the library since it's already hard to support all upstream changes that come from the main Mavlink repository, so adding more stuff is out of the reach.

xinnad, 2022-01-27:
I think with the way that you're doing it is correct. Just misunderstood
how the message worked the first time around. Thanks for your response!