bluenviron/gomavlib #19: Constant overflows for wnum when building for armv7

divyanshu1234, 2022-01-29:
First of all, I'd like to say that this is a really great library. Awesome work by the developers in making something that this well designed and is easy to use.

I used the `dialect-import` tool to generate my own dialect that uses `common.xml`

While building this for armv7 using:
```GOOS=linux GOARCH=arm GOARM=7 go build```

I got the following errors:

```
pkg/dialects/custom/enum_hil_sensor_updated_flags.go:41:5: constant 2147483648 overflows HIL_SENSOR_UPDATED_FLAGS
pkg/dialects/custom/enum_mav_sys_status_sensor.go:75:5: constant 2147483648 overflows MAV_SYS_STATUS_SENSOR
```

This is probably happening because `dialect-import` defines the enum constants as `int` over [here](https://github.com/aler9/gomavlib/blob/795359bc08387f4e99b4c348400413e6315a6071/cmd/dialect-import/main.go#L66). HIL_SENSOR_UPDATED_FLAGS are used as `uint32_t` in messages. This is probably creating problems for arm32 architecture.

divyanshu1234, 2022-01-29:
This can probably be fixed by changing the `int` to `uint32`

aler9, 2022-01-30:
Thanks, this has been fixed.

closed 2022-01-30 by aler9