bluenviron/gomavlib #57: dialect-import error raised 'ERR: unable to open: open custommessages.xml: no such file or directory'

meteoricfarm, 2023-07-31:
Dear contributors,

I'd like to generate custom dialect.
following instruction, I've go error raised like below.

```
ZenBook:~/workspace/drone.dev/gomavlib/message_definitions/v1.0$ dialect-import custommessages.xml 
definition custommessages.xml
ERR: unable to open: open custommessages.xml: no such file or directory
```

has anyone solution for this error?

Thanks for your reply in advance.

Sincerely,

meteoricfarm, 2023-07-31:
FYI, attached the dialect that I used.

```
<?xml version="1.0"?>
<mavlink>
  <!-- XML file for prototyping definitions for common.xml -->
  <include>ardupilotmega.xml</include>
  <version>0</version>
  <dialect>0</dialect>
  <enums>
    <enum name="A_TYPE">
      <description>Detected Anomaly Types.</description>
      <entry value="0" name="A">
        <description>Mechanical failure.</description>
      </entry>
      <entry value="1" name="B">
        <description>Vibration issue.</description>
      </entry>
      <entry value="2" name="C">
        <description>Compass interference.</description>
      </entry>
      <entry value="3" name="D">
        <description>GPS glitch.</description>
      </entry>
      <entry value="4" name="E">
        <description>Power issue.</description>
      </entry>
    </enum>
  </enums>
  <messages>
    <!-- Detected anomaly info measured by onboard sensors and actuators -->
    <message id="43000" name="A_MESSAGE">
      <description>Detected anomaly info measured by onboard sensors and actuators. </description>
      <field type="uint64_t" name="timestamp" units="us">Timestamp (UNIX epoch time)</field>
      <field type="uint8_t" name="a_field" instance="true">whether anomaly has been detected or not</field>
      <field type="uint8_t" name="b_field" enum="A_TYPE">which anomaly has been detected.</field>
    </message>
  </messages>
</mavlink>

```

aler9, 2023-08-07:
Hello, thanks for reporting the issues.

The first issue (unable to open: open custommessages.xml: no such file or directory) has been fixed with #58 and a new version of the library has been released.

The second issue is not an issue: page https://github.com/meteoricfarm/mavlink/blob/master/message_definitions/v1.0/custom_messages.xml is a HTML page that contains the GitHub website, while the correct XML link is https://raw.githubusercontent.com/meteoricfarm/mavlink/master/message_definitions/v1.0/custom_messages.xml

This command works perfectly:

```
dialect-import https://raw.githubusercontent.com/meteoricfarm/mavlink/master/message_definitions/v1.0/custom_messages.xml
```

closed 2023-08-07 by aler9