bluenviron/gomavlib #93: MAVLink XML change to allow exponentiation operator in bitmask flag

hamishwillee, 2024-02-26:
FYI We're modifying the MAVLink XSD validation file in https://github.com/ArduPilot/pymavlink/pull/920 to allow a bitmask value to be declared using the Python exponentiation operator. 

This makes it easier to immediately visualize which bit is set by a flag, because you can use the syntax 
```xml
<entry value="2**15" name="BIT15" />
```
rather than
```xml
<entry value="32768" name=" BIT15" />
```
This is transparent to mavgen because the parser evaluates the number before creating the generated headers. However it may affect other the rust parser
We're not updating the XML yet to allow other parsers time to update.

Any concerns, please raise on https://github.com/ArduPilot/pymavlink/pull/920

aler9, 2024-02-29:
Hello Hamish, thanks for reporting the update and for taking into consideration the effects on this library.

I've updated the library and its unit tests in order to support all enum formats listed inside the PR you mentioned.

closed 2024-02-29 by aler9

hamishwillee, 2024-03-06:
THanks @aler9 . As an aside, this has been and interesting exercise in working out which libraries are most active :-)

PR #94 (2024-02-29): support enum values in binary, hex and exponential format (#93)
Fixes #93