bluenviron/gomavlib #104: OutSystemID should also be allowed as 0

sanderux, 2024-04-21:
I'm writing a system that broadcasts adsb to all systems.
TargetSystemID 0 should be allowed per mavlink standards.

This line does not allow OutSystemID to be set to 0
https://github.com/bluenviron/gomavlib/blob/c2f6b02b4fd10900c8526a8ef0a792f49bb8d3db/pkg/frame/writer.go#L52

@aler9 fyi

sanderux, 2024-04-21:
See here:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L564

aler9, 2024-08-05:
Hello @sanderux, the `OutSystemID` field is unrelated from the `TargetSystem` field - the former is in the frame and identifies the source, while the latter is in the message and identifies the destination. The library doesn't prevent in any way using a `TargetSystem` equal to zero:

```go
node.WriteMessageAll(common.MessageLoggingData{
    TargetSystem: 0,
    TargetComponent: 0,
    ...
})
```

closed 2024-08-05 by aler9