bluenviron/gomavlib #107: Closing node hangs when having multiple serial nodes

JonPichel, 2024-05-08:
Hi,

I've been working on a program to transfer files between two SiK radios. For testing, I open both nodes from the same program and run them on separate goroutines. When both goroutines were done, I've tried closing both nodes, but it hanged on that call and the program didn't return.

First I thought it was a problem with the logic of my FSMs, but I've tried removing that code and the program still has problems closing both nodes.

```golang
package main

import (
	"file-transfer/internal/mavlink"
	"fmt"
	"os"

	"github.com/bluenviron/gomavlib/v3"
	"github.com/rs/zerolog"
)

var (
	logger = zerolog.New(zerolog.ConsoleWriter{Out: os.Stderr}).Level(zerolog.DebugLevel).With().Timestamp().Logger()
)

func main() {
	senderNode, err := gomavlib.NewNode(gomavlib.NodeConf{
		Endpoints: []gomavlib.EndpointConf{
			gomavlib.EndpointSerial{
				Device: "/dev/ttyUSB4",
				Baud:   57600,
			},
		},
		Dialect:     mavlink.Dialect,
		OutVersion:  gomavlib.V2,
		OutSystemID: 10,
	})
	if err != nil {
		logger.Fatal().Err(err).Msg("Creating sender node")
	}
	defer func() {
		fmt.Println("stopping sender node")
		senderNode.Close()
	}()

	receiverNode, err := gomavlib.NewNode(gomavlib.NodeConf{
		Endpoints: []gomavlib.EndpointConf{
			gomavlib.EndpointSerial{
				Device: "/dev/ttyUSB5",
				Baud:   57600,
			},
		},
		Dialect:     mavlink.Dialect,
		OutVersion:  gomavlib.V2,
		OutSystemID: 11,
	})
	if err != nil {
		logger.Fatal().Err(err).Msg("Creating receiver node")
	}
	defer func() {
		fmt.Println("stopping receiver node")
		receiverNode.Close()
	}()
}
```

aler9, 2024-08-05:
Hello, if a node is not closing, it probably means that the serial channel is not closing, i.e. this line is not returning:

https://github.com/bluenviron/gomavlib/blob/b0877b5351e68d3a71d68e682b1df48442b6a374/endpoint_serial.go#L60

This might be influenced by the quantity of bytes you're feeding into the serial port, and consequently, by messages and dialects, but just in terms of size, not content. Dialects have no effect on routines and are just converters that allow to convert Golang structs to bytes and vice-versa.

In order to further debug this issue, you need to provide a list of all open goroutines during the freeze, that can be generated in this way:

1. add this somewhere

```go

// inside the import section
import (
_ "net/http/pprof"
)

// inside main
go func(mux *http.ServeMux) {
    (&http.Server{
        Addr: ":9999",
        Handler: mux,
    }).ListenAndServe()
}(http.DefaultServeMux)
http.DefaultServeMux = http.NewServeMux()
```

2. start the program and wait for the freeze to happen

3. dump the goroutine stack

```
go tool pprof -text http://localhost:9999/debug/pprof/goroutine
```

4. attach here

aler9, 2024-10-27:
closed since no feedback was provided.

closed 2024-10-27 by aler9